V-Squared/v2-Production #1: Shipping Box Design

VillageHubertChen, 2016-04-14:


georgillies, 2016-04-16:
Reusability: Good idea. Unless shipping is much cheaper for folded shipping box and ViCase. Video may be of lower priority.

VillageHubertChen, 2016-04-25:
# Feedback to Georg's comment

Agreed. Meanwhile we had an idea for a → "Reusable Flat ViCase Shipping Box". Please check the added description in the original post. So we may be able to enjoy the best of both worlds.

VillageHubertChen, 2016-05-04:
# Feedback from Xiao Mao

1. By the time the retailer receives the shipment from china, the shipping box will have shipping marks and lots of stickers and packaging tape. Reusing this shipping box will make it look very torn when it arrives to the customer. We should prepare the customer for it so he is not surprised. That said, the shipping box will do its duty just fine, which is to protect the product box and the product. Reusing this box is more environmentally friendly than buying a new shipping box.
2. We need to place a separation sheet between each aluminum sheet to avoid scratches during shipping

closed 2016-05-30 by VillageHubertChen