V-Squared/v2-Production #3: Price List ViCase Mini

VillageHubertChen, 2016-04-27:
ViCase Mini Price List
================

## Price in Shop for ViCase Mini
Production: $ 129 (Once we start shipping we switch from Early Bird to regular)
Early Bird: $ 99 → Shipped in August (offer now to end of June)
Gamma: $ 129 (Gamma Production, Shipping starts June)

## Reference Prices of all ViCase Models to Users in Shop
- ViCase Tiny: $99
- ViCase Mini: $129
- ViCase Small: $149

## Buying Price:
- Ship Now: Gamma: $129  (Shipping Included)
- 2nd Gamma Order 25 pcs: $110  (Shipping Included) (15%)
- 2nd Gamma Order 50 pcs: $99  (Shipping Included) (23%)
- Ship Aug: Early Bird: $99   (Shipping Included) (This is special Early Bird offer)
- Ship Aug: Production 5 pcs: $116  (Shipping Included) (10%) 
- Ship Aug: Production 10 pcs: $110  (Shipping Included) (15%)
- Ship Aug: Production 20 pcs: $103  (Shipping Included) (20%)
- Ship Aug: Production 50 pcs: $97  (Shipping Included) (25%)
- Ship Aug: Production 100 pcs: $90  (Shipping Included) (30%)

## What is included in ViCase Mini
- Aluminum Case including: Screws, Feets, Filter
- V²Panel
- V²Surge
- 2 x SATA Data Cable
- 2 x SATA Power Cable
- ViCase-Cooler-L (Special offer for Gamma Orders only, later is separate)
- Reusable Shipping Box that fits assembled ViPC
- ViCase Access (Special offer only for 1st Sample, all other shipments is separate item)

## Color Options
See open Issues

Orders
=========

## Initial Order:
- Invoice: 21 pcs * $129 / pcs = $2709
- 3 Shipments: 1 pcs, 5 pcs, 15 pcs
- After each shipment you can open issues that we will fix in next shipment
- Turn around time from you publish issue to we ship new production with issues fixed: 20 days

## 2nd Gamma Order

Considerations
==========
- Our costs for small 1 and 5 pcs is higher than $129. But I want to give you opportunity to feedback so you can be sure you can sell ViCase & ViPC
- For these small qty I can not sell you lower than $129. We are not making money on it, but I want to get you going. I hope you join in this spirit: Not make money on selling ViCase, but get it going
- You are very profitable though on selling ViPC: $100 for small configurations, $200 on big configurations. As such you will be motivated to get ViPC going as quick as possible

Price List causing behavior
=================
- You are motivated to get ViPC out ASAP, as this makes most money
- Even you are not making money on ViCase, you are motivated to move out your inventory quickly to place your 2nd Gamma order for which ViCase is discounted and you make more money
- Selling ViCase before August is profitable for you beginning with your 2nd order
- We add free ViCase-Cooler-L for Gamma Samples. This shall motivate you to promote it. Find ways to make this valuable. ViCase-Cooler-L could be best added value to ViCase Mini.

VillageHubertChen, 2016-04-27:
# Feedback from Sascha
All makes sense and sounds good.

bcaswelch, 2016-05-01:
Talk with my parents and are cool with the setup, had a couple of points they want to incorporate

Overall plan-
-I stay in Carbondale and manage advertising, marketing, online shop details
-Shop Headquarters would be at parents house in Nevada, shipment would be sent there.
	-No state taxes for parts, cheaper cost.
	-Much easier with my parents having the logistics done through Nevada (Illinois has terrible taxes)
-Hire father as worker, he would assemble the computers and send them off to people.

So over the summer, we would have normal operations, I would sell from Nevada and show my dad the ins and out of assembling the computers. 

The advantages of this once summer is finished would be me being able to focus on school but still manage the shop. Also the components would be in a safe and secure enviroment (residential house in a nice neighborhood vs. college house that will see plenty of parties over the years. 

Let me know what you think about this plan? Doing it this way would make me feel much more comfortable in terms of managing the shop and school at the same time.

VillageHubertChen, 2016-05-02:
I love it!

VillageHubertChen, 2016-05-02:
# Answering questions from Bobby

## Shipment
- 1 pcs we will use FeDex or cohorts
- bigger qty is with SF Express (Chinese Forwarder)
- Small shipments are about 1 week
- about 10 days SF Air
- about 25 days SF Ship

VillageHubertChen, 2016-05-02:
# Ideas for using the One Golden Sample

1. Arrange for review in Nevada or SF or SJ or LA
2. You can go there by airplane, same day in and out
3. Get several quick reviews done of ViCase and ViPC with the one Golden sample you have, without giving it out of hand

VillageHubertChen, 2016-05-02:
# Ideas on additional Marketing

Bobby has access to Laser Engraving for customizing the ViCase

VillageHubertChen, 2016-05-03:
I moved all ideas to dedicated issues and linked them. Since price list itself is done I am closing this issue :tada:

closed 2016-05-03 by VillageHubertChen