V-Squared/v2-Production #11: M: Design 2 mm Parts 05.10.

VillageHubertChen, 2016-05-03:
- [ ] HC: Write → Features for 2 mm Parts
- [ ] Design Parts according →

VillageHubertChen, 2016-05-03:
# Features for 2 mm Parts

## Quick Snap Top Cover
- New design: One big airflow hole
- 2 mm materials
- magnets on both parts

## Quick Snap Top Filter
- New design: One big airflow hole
- 2 mm materials
- magnets on both parts
- Filter holder squeezes filter between Top Cover and Filter Holder

## Top and bottom cover
- Make two top covers → Black and Red Top Cover
- Miter: 
   - 45° Angle, about 1 mm surface lengths → 0.8 mm deep
   - Both top and bottom cover
- Change to 2 mm strengths. Simplify CNC cuts.
- Top and Bottom cover each is panel form with small parts in cut out of big hole

## No separate small parts
- Move into Top and bottom cover

## Single sided production
- Make all design changes for single sided production
- Housing need to be taller, as we do not cut recess for bracket
- Screw holes in top and bottom frame are countersunk. 
   - If can not cut with CNC, need to do secondary tooling in office with drill press

VillageHubertChen, 2016-05-26:
# Review the new design

- [ ] Make Render Previews of new design and embed into this comment
- [ ] @bcaswelch Review this design with HC. Please see lists of benefits

## Benefits new design
- More consistent design, look and feel. Now top grill matches side grill
- Top filter is more easy to clean
- Cleaning of top and side filter is consistent
- New design allows to fit into manufacturing budget for 129 USD ViCase Mini

## Render Previews

### Standard Front View

### Standard Rear View

### Top Top View

bcaswelch, 2016-05-26:
Hi Hubert,

These changes all seem great! Looks easy to assemble with a consistent aesthetic. 

1-will I be able to VESA mount this unit to a monitor? I definitely want to test out how it looks and feels on a screen.

VillageHubertChen, 2016-05-27:
@bcaswelch Hi, great to hear from you. Yes. We are late and the same time we are early. We changed plan to include all changes for mass production. So the unit that you are getting is final design in all regards, features and details. You will be able to mount to a Display with a free VESA mount in the back. We support 75, 100 and 200 mm grid spacing. However, I believe we forgot to put the VESA bracket in there. I will now check the packaging list for final shipping and add it accordingly. So thanks for asking :-)

VillageHubertChen, 2016-05-27:
@bcaswelch You may be interested in this summary I just wrote:  Revision and Strategy Change Lists → #26

closed 2016-06-07 by VillageHubertChen