V-Squared/v2-Production #5: Tape Out ViPanel & ViSurge

VillageHubertChen, 2016-05-03:
- [ ] Complete Design
- [ ] Trigger review to HC
- [ ] Move documentation to GitHub during review process. Give priority to documentation needed for user to assemble ViCase. Bobby will need it to build ViPC.
- [ ] Prep for factory
- [ ] Trigger HC for final review: Cost, L/T, Production files

VillageHubertChen, 2016-05-17:
# Fix Safety Issue of Fuse Holder

## Problem
After we received all the parts we found that the fuse holder allows for touching the contacts. In case the power cable is still connected to the ViCase this means the user can electrocute himself. 

## Solution
- Remove Fuse from Outside of Case
- Move fuse to inside
- Add Label to warn and explain

## Additional improvements
Solving this problem requires a redesign of a big portion of ViSurge anyways, so we can add other improvements on our backlog list:

- Make Varistor User Replaceable
- Add Spare Fuse and Varistor Holder (Ziplock bag taped to side wall)

bcaswelch, 2016-05-17:
Good catch, definitely an issue that needs resolving. Let me know whenever you upload the snaps and I will make to sure notice this when I get my sample case.

VillageHubertChen, 2016-05-18:
Hi Bobby, we will add the snapshots in a few hours. It would be great if you can review ASAP. This is the last step before putting the PCB to production.

Thanks very much for your appreciation! We were a bit frustrated on this setback, but it is important to us that ViCase is very safe. The new design also adds the very nice feature of intuitive instructions on a label and user replaceable spare parts. To my knowledge this is the first surge protector that the user can repair himself after a powerful surge burned it out.

VillageHubertChen, 2016-05-18:
@bcaswelch . I uploaded the images for the new design. Please review. We can not work tomorrow due to construction. I will check in the day after tomorrow.

bcaswelch, 2016-05-18:
Hi Hubert, these snapshots look great! Seems like a much safer alternative while also keeping the vicase stylish. I think the text will be very helpful, and anyone trying to change the fuse will most definitely read it.
-Changing the fuse is a feature a would highlight, because I could see a lot of people not knowing you supply everything needed and going out to a store to try and fix it.
-Is it ok to keep a plastic bag in a hot computer case? I think its a great location for storage and if its fine in there, no need to change?

VillageHubertChen, 2016-05-22:
Hi Bobby, thanks for the feedback.

1. I agree we should highlight in the user documentation or even marketing materials that we include even the user replaceable parts. Lets work on that when we make the product web site. You can start thinking about how to market it
2. The air inside the housing is not getting so hot as to affect the plastic bag.

I take your comments than as approval and we will tape out the PCB today. We could not work the last few days because we had construction in the building that was unbelievably loud.

bcaswelch, 2016-05-22:
1. Yes, I approve of the changes. Self repair is a feature many want / appreciate with computers so this will be a positive message to promote.

2. Okay good, figured it wasnt going to be an issue but thought I'd bring it up just in case. I think its great to keep inside the case because small bags that come with computers ALWAYS get lost. I wonder if the bag could include extra screws/mounting brackets..? 

I'll try to check my email periodically for updates and reply as soon as possible. In other news my LA class has been going wonderfully so far :)

VillageHubertChen, 2016-05-22:
1. Yep. We should have one Article only focused on all design features regarding self repair. We should promote iFixit and by the time we publish it we should try to get them cross link it. That should help the spirit of iFixitr to have a manufacturer designing a product following iFixit guidelines and bring us some traffic.
2. Great idea on adding spare screws. We will do this right away. 
3. Change of topic: I am realizing that we are spending a huge amount of time on the surge protection and power distribution. It is very, very difficult to make it this small and still follow all safety regulations. At the same time when using ViCase in real life, this makes a big difference in removing the cable clutter that comes with a PC. We need to find a way to bring this message across to get a Return of Investment. This is a very unique selling point :-)

I can not use email, gmail is blocked in China. Is it possible you call me on skype. I really want to talk to you. I am currently spending a lot of time to complete the planning and organization part of the V²Site and would be grateful to double check with you if this is the best use of my time at this moment to prepare selling ViCase.

VillageHubertChen, 2016-05-22:
we updated the image [Bag with Spare Parts](#bag-with-spares) as per your suggestions and added spare screws. Note: We are not using any spare brackets that might get lost.

closed 2016-05-30 by VillageHubertChen