V-Squared/v2-Production #40: Video Review (0) V²Shop-Midwest

VillageHubertChen, 2016-05-31:
## Complete Design Review all Features ViPC / ViCase

### Preparing Video Conference
- [ ] Set up recording of entire conference
- [ ] Create a complete list of all features of ViPanel, ViSurge, ViCase-Access

### Video Conference

Show the following features (not yet a complete list)

- [ ] ViPanel PCB and its functions
- [ ] ViSurge PCB and its functions
- [ ] ViCase-Access PCB and its functions
- [ ] Show all cables and their functions
- [ ] Show a complete assembled ViPC
- [ ] Show all Assembly / Configuration Options
   - [ ] Power Options
      - [ ] Flex ATX (300W, fan)
      - [ ] pico ATX (120W, fan-less)
      - [ ] micro ATX (240W, fan-less)
   - [ ] CPU Cooling Options
      - [ ] ViCase Cooler L
      - [ ] ViCase Cooler XL
   - [ ] Top Cover Options
      - [ ] Black
      - [ ] Red
   - [ ] F³ = Fluid-Form-Factor Options
      - Mount to Wall
      - AiO1: Mount behind VESA Display
      - AiO2: Standing behind VESA Display
      - Mini PC
    - [ ] ViCase Access
    - [ ] ViCase-Remote
    - [ ] ViCase-Chroma (Programmable Color Lights)
    - [ ] Disk mounting options

### Post Video Conference
- [ ] Sign off or create Engineering Change List
- [ ] Discuss if we should edit and publish this video to get viral marketing going?
- [ ] How and where do we publish this Video. If not for Viral marketing, V²Shop-Midwest need access to it
- [ ] Create an Issue where we are documenting these products in new Products Pages
- [ ] Discuss Promotion Option:
   1. Create Product Pages in Site and Shop with pricing for all Accessories
   2. During Community Promotion period all accessories included

VillageHubertChen, 2016-06-01:
# Discussion on Silent and 0dB PC

- [ ] Ultra Low Fan speed option
- http://www.amazon.com/HDPLEX-Internal-Power-Adapter-Output/dp/B00QSRBPB2
- http://www.amazon.com/Original-PA-9E-Replacement-adapter-Notebook/dp/B00AVMYMUG
- http://www.amazon.com/HDPLEX-DC-ATX-Power-Supply-16V-24V/dp/B00KVWHUXW
- www.hd-plex.com (check if they have a bigger than 160W option, this site is blocked in China)

bcaswelch, 2016-06-02:
Agree with the HD-Plex power supplies, seem like the best option in terms of space and noise. People won't need more than 250w for a SFF computer.

bcaswelch, 2016-06-02:
Will you have an AiO converter to show?

VillageHubertChen, 2016-06-02:
# Answering bcaswelch

1. **HD-Plex** we are designing them in. It is very hard work because it is really tight and cooling airflow is complex.
2. **AiO converter:** I assume you mean the "ViCase-Stand", that is the Display Stand that also holds ViCase, Keyboard and Mouse? I have added it to the list. Thanks for the reminder.

closed 2016-06-06 by VillageHubertChen