V-Squared/v2-Production #44: Create Engineering Change List

VillageHubertChen, 2016-06-06:
# Action
- [x] Do →  Video Review (0b) V²Shop-Midwest #43 
- [x] Do →  Video Review (0) V²Shop-Midwest #40 
- [ ] ML: Detailed review of Design
- [x] ML, HC: Detailed joined review of new Front Panel Design
- [ ] BC, TC, KC: Report any changes in comments no later than: June 7th
- [ ] HC: Approve all changes
- [ ] ML: Implement all changes

# ECL
- [ ] New Front Panel Design to accommodate HDPLEX PSU
- [ ] Add three holes total for DC Jack for picoATX / nanoATX PSU. The whole size we have is in the middle. Place one left with 2 mm small diameter and one to the right with 2 mm larger diameter. This allows to use different kinds of DC-Jack. We do this because the DC-Jack diameter is not standardized.

bcaswelch, 2016-06-08:
**Possible changes**

1. Make changes to dust filter connector. Use magnetic strip to connect to cover? Many cases currently do this and is well received. 

2. Is the bottom cover easily removable? This could be important for users to replace hard drive. 

3. Braces for the outer shell (less squishy).

bcaswelch, 2016-06-08:
Dear Bobby,

First of all thank you for compiling the Engineering Change List.

Unfortunately this is not the right place. This is: →  Create Last Engineering Change List #44 

Remember I introduced it in the Sprint Kick Off meeting today. It is in the column: "This Sprint" of https://waffle.io/V-Squared/V-Squared.github.io 

Please do the following:

- [ ] Copy paste your original text as new comment to →  Create Last Engineering Change List #44
- [ ] Copy and paste this answer as second comment
- [ ] Answer the following questions: (in a third comment)

1. **What is a Dust Filter Connector?** Do you mean the holding mechanism? If so I am confused, because I showed you that the Dust Filter Holding Frame is clamping the Dust Filter with magnets. Please → https://github.com/V-Squared/v2-Production/issues/11#issuecomment-221780126
2. **Bottom Cover is not easily removable:** Remember, we talked about this one too. Instead we plan to make the motherboard easy removable. You need to open 4 screws and you can remove the motherboard. This however will only work if you choose a motherboard where the cable connectors are suitable and if you choose a cable wiring, so that the cables are long enough. This is not an ideal solution. If we were to make the Bottom Cover easy removable, we would make the housing significantly more complex and more expensive, and assembly will take longer. I am open to talk about this again tomorrow. Please make sure Tom is in that call, as he is the one who need to assemble the PCs. If we were to add this feature, we will add to the assembly time. I feel that the proposed solution is sufficient. We have so many plus points that we can afford a minus.
3. **Braces for the out shell:**
   1. What do you mean by braces? Carry Grips? 
   2. What do you mean by squishy? The fluffy nature of the side grill?
   3. What is the purpose of the braces? Where do they go?

Please note that after you move this I need to spend some time to archive this issue, which is not as straight forward as you think. Please be careful with creating new issues. Please spend the requisite time to check of an issue already exists on the topic.

bcaswelch, 2016-06-08:
1. The way the dust filter sticks onto the metal plate. Most cases use a magnetic strip that will hold onto the metal, this could be a more straightforward way of clamping it to the case.

2. Thats fine,  this is going to be a fairly complex case to build anyway so no need to make it even more complicated. Not having easy access to the bottom will be mitigated by having very open access to the sides and top.

3.Braces that would be inside of the case lining the wall so that when you press on the sides, the squishy material doesn't flex as much. The purpose would be to give the sides more sturdiness. But I think at this point theres enough parts to the case that there doesn't need to be a significant change.

VillageHubertChen, 2016-06-08:
# Brief of Skype Call

1. This design is accepted by Bobby. It is more expensive as magnetic strips, but faster to use. Gluing strips to filter is no good, as the filter would be no longer machine washable. This design is more expensive, but also higher quality, which is in match with target audience. V² Compatible Case vendors certainly have an opportunity to safe cost here when making a consumer quality case :-)
2. Just a note: Keep thinking on how to make access of disks easier. Each time you are handling the ViCase you can keep thinking for an easy design that will allow this. We will be excited to implement it. We just have not found an easy way yet to do it.
3. Noted. We are looking for a similar thing. I suggest get the first ViCase in hand and see if it is an issue. If it is, lets brainstorm on how to fix it.

VillageHubertChen, 2016-06-08:
# Please confirm this is V²Shop Design Sign Off

@bcaswelch Please confirm in a comment that this is your official sign off. From this moment on any feature request goes into the next design spin, which would be the 20 pcs shipment R1. Please post comment: V²Shop-Midwest signs off ViCase Golden Sample Design"

bcaswelch, 2016-06-10:
I sign off