V-Squared/v2-Production #47: M: CNC-Cut 2 mm Fixture

VillageHubertChen, 2016-06-12:
CNC cut fixture to hold 2 mm parts

VillageMichaelLin, 2016-06-12:
The link of the trello card for this fixture: https://trello.com/c/wTKSog5L/543-fixture-for-cutting-2-mm

VillageMichaelLin, 2016-06-12:
- [x] Write the trello card for this fixture
- [ ] Choose the CNC shop
- [ ] Buy the things need for the fixture
- [ ] Cut the fixture in the CNC shop
- [ ] Assembly the fixture

VillageHubertChen, 2016-06-12:
Dear Michael. Please check how I moved the content for the task list to the top comment and how I edit the trigger of you to me in your comment. Please check with me after kick off meeting in case I have not reviewed the Trello Card by 3 pm.

VillageMichaelLin, 2016-06-13:
@VillageHubertChen Please review → [Trello►How to choose the CNC shop](https://trello.com/c/zKO3eb88/545-how-to-choose-cnc-shop)

VillageHubertChen, 2016-06-14:
@VillageMichaelLin I reviewed and edited → [Trello►How to choose the CNC shop](https://trello.com/c/zKO3eb88/545-how-to-choose-cnc-shop). Please see the comment.