V-Squared/v2-Production #48: M: Design change to fit DC-Jack-PCB for HDPLEX 250W DC-DC

VillageHubertChen, 2016-06-16:
- [x] Document problem → see below
- [x] Brainstorm if this problem can be solved → yes, see below
- [x[ determine joint mounting hole Ø → 8.5 mm
- [ ] Draw DC-Jack-PCB for HDPLEX 250W DC-DC
- [ ] Place DC-Jack-PCB for HDPLEX 250W DC-DC into Issue to get V²Shop-Midwest to double check the dimensions
- [ ] Change Rear Panel Design
- [ ] Change all Production documentation for 2 mm part

# Problem to solve
The DC Cable jack is not a screw in but a very big PCB:

[image]

VillageHubertChen, 2016-06-17:
# Guessed PCB design, showing to tight fit

[image]

VillageHubertChen, 2016-06-17:
# List of Questions to HD-Plex

- [ ] Do they ship the small or big version of DC Jack PCB, see pictures above
   - do they have two options available?
   - can they do the solder in version for us?
- [ ] Why shipping takes so long
- [ ] What is average lead time on buying Power Supplies from HD-Plex? How much inventory do they usually keep?

bcaswelch, 2016-06-17:
Good news- The small type is the one I received.

VillageHubertChen, 2016-06-18:
@bcaswelch Great! However we still must have the answers to the questions above: [List of Questions to HD-Plex](#issuecomment-226680737)

bcaswelch, 2016-06-18:
Email reply:

Hi Robert

1. You mean the 7.4*5.0 PCB for DC input?  Yes this is soldered onto PCB for mounting purpose since there is no self lock DC input for 7.4*5.0.

2. No,  this delay is caused by waiting for the the new version of 160AC-DC which is much more robust. Future orders will be shipped within 24 hours. 
3. I expected to have more than 100pcs of 160AC-DC and 250W DC-ATX starting from next month.  
Larry

VillageHubertChen, 2016-06-18:
@bcaswelch Great! Did you informed them on our project?

bcaswelch, 2016-06-18:
No, I can reply to him and give details on it. Say that we are developing a computer case and plan to integrate hdplex power supply dimensions?

VillageHubertChen, 2016-06-18:
I think it is needed. He said he got the new 160W. It is great that it is better. But what if it is different? The one you got is from Amazon and old version or is it from HDPlex and new version?

In any event it would be good if they could check our drawings to see if they know they will make changes to some dimensions within this year. Knowing that before hand we can change the ViCase accordingly. Otherwise you may face the problem that next shipment of their PSU will not be possible to mount in ViCase.

bcaswelch, 2016-06-18:
Ok, I'll ask them about it. I got the new one and it appears the same, I guess just with higher quality parts. Thats why it took so long to ship it out.

VillageHubertChen, 2016-06-18:
OK. Got it. I think we now have all information to complete this design. Thanks for all the follow up. You removed the gating from this issue. I will still keep the label until michael can complete the design.

VillageHubertChen, 2016-06-18:
@VillageMichaelLin Bobby and me believe we have answered all your open questions in the conversation above. Please check it out and complete the design. Please update → [Issue > V-Squared > Evaluate HD-Plex PSU #132 ](https://github.com/V-Squared/V-Squared.github.io/issues/132) accordingly.

bcaswelch, 2016-06-18:
Sounds good.

VillageHubertChen, 2016-06-18:
@bcaswelch Then your job is done here. We take over from here. Thanks so much!

closed 2016-06-27 by VillageHubertChen